makerbase-mks/MKS-GEN_L #1: Dual Z problem

joydi, 2020-01-01:
Hi,

i have an problem with  the MKS gen l v2.

I want to build an core xy with two z stepper driver i have the tmc2208 on the board.

When i  do the following in the configuration.h

#define X_DRIVER_TYPE  TMC2208
#define Y_DRIVER_TYPE  TMC2208
#define Z_DRIVER_TYPE  TMC2208
//#define X2_DRIVER_TYPE A4988
//#define Y2_DRIVER_TYPE A4988
#define Z2_DRIVER_TYPE TMC2208
//#define Z3_DRIVER_TYPE A4988
#define E0_DRIVER_TYPE TMC2208

and in the configuration_adv.h

#define Z_DUAL_STEPPER_DRIVERS
#if ENABLED(Z_DUAL_STEPPER_DRIVERS)
  //#define Z_DUAL_ENDSTOPS
  #if ENABLED(Z_DUAL_ENDSTOPS)
    #define Z2_USE_ENDSTOP _XMAX_
    #define Z_DUAL_ENDSTOPS_ADJUSTMENT  0
  #endif
#endif

at the compiling stage the error 

"TMC2208 or TMC2209 on Z2 requires Z2_HARDWARE_SERIAL or Z2_SERIAL_(RX|TX)_PIN."

comes. 

Can you help me ?

Thanks

makerbase-mks, 2020-01-11:
Hi, you need pins_MKS_GEN_L_V2.h file, add:
#ifndef Z2_SERIAL_TX_PIN
  #define Z2_SERIAL_TX_PIN    20
#endif
#ifndef Z2_SERIAL_RX_PIN
  #define Z2_SERIAL_RX_PIN    21
#endif

sjasonsmith, 2020-02-01:
@makerbase-mks, the AVR SoftwareSerial library requires the pin to support change interrupts, which pin 21 does not.

This results in a communication error when used with Marlin. Can you confirm whether the configuration shown above has actually been tested to work?

I can work around the issue by using a different pin on the board, but that requires added "flying leads" with is contrary to the claims advertised for the board.

makerbase-mks, 2020-02-14:
1. This is indeed a hardware design flaw, and we will improve it in the new version.
2. If you have E1 UART mode, you can solder the D12 signal to the silkscreen side of R37 resistor.
[image]
You need pins_MKS_GEN_L_V2.h file, add:
#ifndef Z2_SERIAL_TX_PIN
#define Z2_SERIAL_TX_PIN 20
#endif
#ifndef Z2_SERIAL_RX_PIN
#define Z2_SERIAL_RX_PIN 12
#endif

closed 2020-04-07 by makerbase-mks

borland1, 2020-04-14:
The suggested jumper and soldering would be very difficult to do.  Because each TMC2208/TMC2209 one wire serial ( PDN_UART ) utilizes up to four addresses, it would be much easier set separate addressing for each driver module (MS1, MS2), then use the same PDN_UART pin for both Z1 and Z2 motor drivers.

This could be accomplished with one wire dupont jumper between ZUART and E1UART.   Then set Z2 driver module by installing single jumper at MS1 to VDD (+5V), while Z1 driver module has no jumpers on MS1 or MS2.

Then in Configuration_adv.h file, assign address for Z2 like this:
#define Z1_SLAVE_ADDRESS 0  // default value
#define Z2_SLAVE_ADDRESS 1 // set with MS1 pulled high

sjasonsmith, 2020-04-14:
> Because each TMC2208/TMC2209 one wire serial ( PDN_UART ) utilizes up to four addresses

This is true for TMC2209 drivers, but TMC2208 do not support addresses; they are always zero.

For 2208 you will need to assign a new RX pin as the MKS post suggested. With Marlin 2.x you can now define the same pin for RX and TX, which would allow you to use a single jumper wire from the servo connector to the E1UART connector, with no soldering required,

AndresCidoncha, 2020-05-16:
Hey guys, I tried @sjasonsmith option with Marlin 2.x and seems to work fine, but I have doubts about one thing:
```
		X	Y	Z	Z2	E
Address		0	0	0	0	0
Enabled		false	false	false	false	false
Set current	1200	1200	1200	1200	1100
RMS current	1160	1160	1160	1160	1049
MAX current	1636	1636	1636	1636	1479
Run current	20/31	20/31	20/31	20/31	18/31
Hold current	10/31	10/31	10/31	10/31	9/31
CS actual	10/31	10/31	10/31	10/31	9/31
PWM scale	12	12	12	12	11
vsense		0=.325	0=.325	0=.325	0=.325	0=.325
stealthChop	true	true	true	true	true
msteps		16	16	16	16	16
tstep		max	max	max	max	max
PWM thresh.	98	98	660	19	59
[mm/s]		100	100	3	104	30
OT prewarn	false	false	false	false	false
off time	4	4	4	4	4
blank time	24	24	24	24	24
hysteresis
 -end		2	2	2	2	2
 -start		1	1	1	1	1
Stallguard thrs	0	0	0	0	0
DRVSTATUS	X	Y	Z	Z2	E
sg_result	0	0	0	0	0
```

As you can see, seems to be working, but there are 2 differences between the Z and Z2 values:
```
                         Z      Z2
PWM thresh.	660	19
[mm/s]		3	104
```

It's this ok? Shouldn't I have the same value for both drivers?

ysnuysl, 2021-11-02:
> This is true for TMC2209 drivers, but TMC2208 do not support addresses; they are always zero.
>
> For 2208 you will need to assign a new RX pin as the MKS post suggested. With Marlin 2.x you can now [...]

**Hi sjasonsmith, "With Marlin 2.x you can now define the same pin for RX and TX, allowing you to use a single jumper wire from the servo connector to the E1UART connector without the need for soldering." I don't fully understand this sentence.
as suggested by makerbase; "pins_MKS_GEN_L_V2.h file, add:
#ifndef Z2_SERIAL_TX_PIN
#define Z2_SERIAL_TX_PIN 20
#endif
#ifndef Z2_SERIAL_RX_PIN
#define Z2_SERIAL_RX_PIN 12
I added #endif". Now where should I make a single jumper wire from D12 to E1UART connector without soldering?

ysnuysl@gmail.com**

Gixie69, 2023-07-29:
Im trying to duo Z my Nano V1.2. I have done the Uart but as my drivers are BTT 2209 they have the double ended pins to put jumper cables on. Annoyingly they plug into the WIFI socket on the board though. Wasted money on the wifi module there.  Trouble is, I need a second place to plug in the Uart for the second Z as its in the E0 position on the board Does it plug in where the original E0 went? and where does the one for the Z2 Uart go???

Gixie69, 2023-07-29:
On the Robin Nana v1.2, The Wifi socket Green/Black sides, Do the board pins match the stepper driver pin lay out???

daveysir, 2023-08-03:
> This is true for TMC2209 drivers, but TMC2208 do not support addresses; they are always zero.
>
> For 2208 you will need to assign a new RX pin as the MKS post suggested. With Marlin 2.x you can now [...]

Thank you for all the information. I have been fighting to get this working. I did set the pins to 
#ifndef Z2_SERIAL_TX_PIN
#define Z2_SERIAL_TX_PIN 20
#endif
#ifndef Z2_SERIAL_RX_PIN
#define Z2_SERIAL_RX_PIN 12
And put a jumper wire from pin 12 to the left E1UART pin, but i get no result so far. 
I have the MKS GEN L v2.0 board and 2209 steppers in Standalone mode.

I hope you guys have a clue on what to do. Thanks

daveysir, 2023-08-03:
> The suggested jumper and soldering would be very difficult to do. Because each TMC2208/TMC2209 one wire serial ( PDN_UART ) utilizes up to four addresses, it would be much easier set same/separate addressing for each driver module (MS1, MS2), then use the same PDN_UART pin for both Z1 and Z2 motor drivers.
> 
> This could be accomplished with one wire dupont jumper between ZUART and E1UART. Then optionally, set Z2 driver module by installing single jumper at MS1 to VDD (+5V), while Z1 driver module has no jumpers on MS1 or MS2.
> 
> Then in Configuration_adv.h file, assign address for Z2 like this: #define Z1_SLAVE_ADDRESS 0 // default value #define Z2_SLAVE_ADDRESS 1 // leave at 0, or optionally set with MS1 pulled high

Thanks for this info. I also tried this method ( as far as i could understand it ) And soldered a wire from MS1 to VDD on the stepper for the E1 (Z2) and on the board i used a dupont wire from The left E1UART pin to the left ZUART pin. In the firmware i set the pins to 
#ifndef Z2_SERIAL_TX_PIN
#define Z2_SERIAL_TX_PIN 20
#endif
#ifndef Z2_SERIAL_RX_PIN
#define Z2_SERIAL_RX_PIN 21

I use 2209 steppers in standalone mode. I also tried setting the z2_serial_RX_pin to the same as the Z. and i tried the same for the TX pin. This resulted in the second z being very slow.
I can't get it to work. hope someone can help me get on the right track

daveysir, 2023-08-03:
As a last resort i cut the bottom pin of the pdn (RX) and added a jumper wire on top to pin 12. As aspected.... No result. Because i don't know what i'm doing :'(