makerbase-mks/MKS-GEN_L #8: Sensorless parking tmc2209 on Z does not work

lonelymyp, 2020-09-07:
I have configured all 3 axes of the printer for sensorless parking for marlin 2.0.
The parameters were adjusted strictly according to the video instructions, all 3 axes are identical.
The X and Y axes are parked without problems and the Z axis does not trigger. 
No signal appears on the contact Zdiag. It seems that the driver is simply not giving the desired signal. Changed the driver in places, did not help.

Maybe something else needs to be adjusted for the z axis?

lonelymyp, 2020-09-07:
It looks like if the axle parking speed is higher than HYBRID_THRESHOLD for that axle then no parking signal is generated.
HYBRID_THRESHOLD must be higher than the parking speed for parking to work correctly.

makerbase-mks, 2020-09-08:
So, sensorless homing is works?

lonelymyp, 2020-09-08:
Yes, after setting HYBRID_THRESHOLD

makerbase-mks, 2020-09-14:
I will close it

closed 2020-09-14 by makerbase-mks