makerbase-mks/MKS-GEN_L #14: MKS GEN_L V2.1 TMC2208 TMC connection error

BigUhg2, 2022-06-03:
Marlin bugfix-2.0.x
echo: Last Updated: 2022-06-01 | Author: (Alan Fereday, TMC2208)
echo: Compiled: Jun  3 2022
echo: Free Memory: 4249  PlannerBufferBytes: 1248
echo:V86 stored settings retrieved (629 bytes; crc 47652)
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW

Followed tutorial, still having this error, What is next?

Sourdough75, 2023-11-23:
I am also having a similar issue using both tmc2208's and tmc2209's
Connecting...
echo:start
Marlin 2.0.7.2
echo: Last Updated: 2020-10-15 | Author: (thisiskeithb, Ender-3)
echo:Compiled: Nov 21 2023
echo: Free Memory: 2785  PlannerBufferBytes: 1200
DIGIPOTS Loading
DIGIPOTS Loaded
DIGIPOTS Loading
DIGIPOTS Loaded
echo:V82 stored settings retrieved (657 bytes; crc 16010)
Testing X connection... Error: All LOW
Testing Y connection... Error: All LOW
Testing Z connection... Error: All LOW
Testing E connection... Error: All LOW
Printer is now online.

dMarabou, 2023-11-23:
Wich manufacturer? 
U have to connect them properly if its Not MKS's TMCs.

udaravima, 2024-05-29:
Testing X connection... OK
Testing Y connection... OK
Testing Z connection... OK
Testing Z2 connection... Error: All LOW
anyone have comment on this?
checked on swapping TMCs and motors also, No luck
firmware problem?
Marlin\src\inc\Warnings.cpp:206:6: warning: #warning "Note: Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)" [-Wcpp]
     #warning "Note: Auto-assigned Z2 STEP/DIR/ENABLE_PINs to unused En_STEP/DIR/ENABLE_PINs. (Define NO_AUTO_ASSIGN_WARNING to suppress this warning.)"
     is this error anything to do with this?